barlowtj48/MIDI-Stepper-Motor-Music #4: Problem with 1 channel of CNC shield

DavidShoe, 2024-01-02:
Configured everything (I think), and when I run the sample FL project 3 of the channels (X,Y,Z) all seem to be working as expected.  But the A channel doesn't turn the stepper. 

I have swapped the driver module / cables and motor and it seems to be an issue with the A channel configuration.  Is there anything different I need to do with it?

This is a Mega 2560 using a CNC shield and 4 A4988 drivers.

levfo, 2024-01-11:
Having the same issue! Nothing from Channel A

barlowtj48, 2024-01-15:
Sorry for the delay. I will be updating the instructions to note the extra pins that are needed to enable the A stepper driver "direction" and "step" output.

barlowtj48, 2024-01-15:
Added documentation for this issue. Tested and confirmed that without the leads plugged in the "A" driver fails to output. 
New photo can be found in [Hardware Assembly Section](https://github.com/barlowtj48/MIDI-Stepper-Motor-Music?tab=readme-ov-file#hardware-assembly)

closed 2024-01-15 by barlowtj48