barlowtj48/MIDI-Stepper-Motor-Music #6: loopMIDI alternative for Mac or Windows ARM?

Rocky56gh9, 2024-01-05:
Hi - awesome project! I'm running an M2 Mac with Windows 11 ARM installed, and loopMIDI is incompatible with that version of Windows. Do you happen to have any thoughts on alternatives that would work on ARM, either Windows or Mac? Thanks!

Rocky56gh9, 2024-01-05:
I found this, but have no idea where to start if this is even viable.

I just recently got into some home projects for displaying various content sources on epaper screens. Your project looks like a lot of fun but like you suggested, don’t want to commit if there is a fundamental blocker. 

https://alternativeto.net/software/hairless-midi--serial-bridge/about/

EJEmmett, 2024-01-05:
From what I can see, Mac also has their own software to do MIDI routing ([MIDI Studio](url)). 

There's also [SoundFlower](https://github.com/mattingalls/Soundflower) but it doesn't support M1 chips and hasn't been updated since 2014. I can't verify anything as I don't have a Mac but still worth a look. 

[MIDIFire](https://audeonic.com/midifire/) looks like it does virtual routing too but costs $11.99.

Rocky56gh9, 2024-01-05:
And finally got to the end and saw, "Add Mac instructions" on the code page, so that may be my answer!

barlowtj48, 2024-01-15:
> Hi - awesome project! I'm running an M2 Mac with Windows 11 ARM installed, and loopMIDI is incompatible with that version of Windows. Do you happen to have any thoughts on alternatives that would work on ARM, either Windows or Mac? Thanks!

@Rocky56gh9 That's an interesting setup you've got there 😄. I looked around to see if loopMIDI could be rebuilt for different architectures and because it's an old closed source project I don't think there's any chance with that. The "Add Mac instructions" was applicable to users' who were still using MacOS, where there is a built in MIDI virtual output that can be used by the script. 

Because you're running Windows on arm64 I don't think there's much that can be done for live output. Regardless, there is a way that you can modify the python script so that instead of opening up a MIDI port, it just opens a single midi file. I've never tested this but believe it might be a workaround. If you're set on using Windows I wouldn't proceed with buying things to make this project, though I've confirmed this setup works fully-featured when using MacOS. Let me know what you think!

Rocky56gh9, 2024-01-15:
Sorry, I was unclear. I’m running MacOS, and in addition I have Win11 ARM virtualized on the same machine. LoopMIDI doesn’t work on that version of Windows. It sounds like I should instead be able to follow the Mac instructions, no?

Rocky56gh9, 2024-01-19:
> @Rocky56gh9 That's an interesting setup you've got there 😄. I looked around to see if loopMIDI could be rebuilt for different architectures and because it's an old closed source project I don't [...]

@barlowtj48 , to confirm I am interpreting the Readme correctly, the section about loopMidi and the IAC driver are interchangeable? LoopMidi for Windows and IAC for Mac? If so then I think I’m going to try this!

barlowtj48, 2024-01-20:
@Rocky56gh9 That's perfect! If you're set up to easily switch between the two then I'd send it. The notes that were added actually came from someone else that added those in, and I've since confirmed that the project works well (live) on MacOS and Windows.

closed 2024-03-01 by barlowtj48