barlowtj48/MIDI-Stepper-Motor-Music #13: Broken on W11 24H2

TehH4rRy, 2025-01-26:
Not sure if this is a LoopMidi problem or what, but I was forced onto W11 24H2 (Thanks Microsoft) and now whenever I play MIDI files they just come out my PC speakers. 

The script detects the midi port and my arduino fine. Reinstalled LoopMidi...not sure what else to try?

barlowtj48, 2025-01-26:
When you are playing the MIDI files, what software are you using?

barlowtj48, 2025-01-27:
Interesting. My understanding of windows media player is slim at best, besides knowing it is very very outdated. I would recommend looking around for some other freeware MIDI editors to use that you can more easily configure the files rather than just a one-shot playback. I know that's not too helpful but non-configurable software that stops working doesn't leave a lot of room for fixes to be made.

TehH4rRy, 2025-01-28:
Ah I tried a few different programs to play MIDI but they just come out my speakers, even rolled back the 24H2 Update as well. The driver in LoopMidi is unchanged. Seems it's broken the way MIDI is routed to LoopMIDI. 

Oh well, I had a bit of fun with this project. I've got the hardware, guess I find one which works on Linux.